stevearc/vim-arduino #1: Property line is invalid

binaryplease, 2016-05-09:
I get this error when trying to copile/upload.

The quotes look strange (' and ")

[image]

stevearc, 2016-05-10:
Hi binary,

What do you get when you do a `:echo g:arduino_programmer`? It should say `arduino:usbtinyisp`.

Assuming that looks normal, you can add this to `vim-arduino/autoload/arduino.vim` on line 264:
```
exe ":silent !echo " . cmd
```
This will print out the full command that it's running, which should give you some clues.

binaryplease, 2016-05-10:
Thanks!

The :echo g:arduino_programmer looks normal.

I added "exe ":silent !echo " . cmd" as you recommended, but the messages are cleared to fast to see the output properly. If I saw it right, the error is the same though.

stevearc, 2016-05-11:
With that line you're not looking for the error, you're looking for the line that's actually running the `arduino` command, so you can check the args and run it manually if need be. It will be in your terminal history if you just scroll up, or quit vim and scroll up. If you want it to stop immediately on printing you can remove the `silent`

closed 2020-12-11 by stevearc