stevearc/vim-arduino #2: error while loading shared libraries

azamatktl, 2016-06-15:
When I try to upload *.ino file:

/home/azamat/arduino-1.6.9/hardware/tools/avr/bin/avrdude: error while loading shared libraries: libtinfo.so.5: cannot open shared object file: No such file or directory

How can I solve this.
Please, help!

stevearc, 2016-06-15:
Do you get the same error when using the arduino IDE? I'd suggest googling around and fiddling with your libraries/installed packages. At a glance, it seems like other people have had similar problems [here](http://arduino.stackexchange.com/questions/21309/error-while-loading-libtinfo-so-5-solved) and [here](https://forums.gentoo.org/viewtopic-p-7815500.html?sid=f67283068e3655b2f9e259a2097efef6)

azamatktl, 2016-06-15:
Thanks stevearc. The problem was that Arduino IDE needed the older version of libtinfo.
Solved this by symlinking: 
cd /lib
ln -s libtinfo.so.6 libtinfo.so.5

closed 2016-06-15 by azamatktl