stevearc/vim-arduino #3: missing Arduino boards list

rikri, 2016-11-05:
Hi,
maybe I do something wrong, but I'm not able to choose different board than default Uno...
I have thought that the list is generated automatically but I see none:
[image]

If I write any number I always get this error:
[image]

Should I (somehow) input available boards manually?

stevearc, 2016-11-05:
vim-arduino attempts to load the boards from the same place that the Arduino IDE stores them. The error that you're seeing means that it can't find the board files for some reason. I have a few questions that might help me diagnose/reproduce the problem:

* What OS are you using?
* What version of Arduino IDE are you using?
* Look in your arduino directory (mine is in `/usr/local/share/arduino/`) under the `hardware` directory. Are there any files anywhere under there named `boards.txt`? For example, I have one at `/usr/local/share/arduino/hardware/arduino/avr/boards.txt`
* What is the output of `which arduino`?
* Inside of vim, what is the output of `:echo arduino#GetArduinoDir()`?

rikri, 2016-11-05:
- What OS are you using?
_Arch Linux_
- What version of Arduino IDE are you using? _1.6.11_
- Look in your arduino directory (mine is in /usr/local/share/arduino/) under the hardware directory. Are there any files anywhere under there named boards.txt? For example, I have one at /usr/local/share/arduino/hardware/arduino/avr/boards.txt
_Mine is in /usr/share/arduino._
_I have tried to add "let g:arduino_cmd = '/usr/share/arduino/arduino'" into my .vimrc but it didn't help._

- What is the output of which arduino?
_/usr/bin/arduino_
- Inside of vim, what is the output of :echo arduino#GetArduinoDir()?
_/usr/bin_

Thank you for your help!

stevearc, 2016-11-05:
Okay, I think I figured it out. I pushed a new commit to master which should have a fix for you. All you need to do is add this to your `.vim/ftplugin/arduino.vim` or `.vimrc`:

```let g:arduino_dir = '/usr/local/share/arduino'```

The problem was that I was searching for the arduino dir using the `arduino` executable found in the PATH. But that assumes that the file in the PATH is either in its original directory or is a symlink to the executable in that directory. If it was moved or copied into the PATH then we would fail to find the correct root directory for the arduino files.

rikri, 2016-11-05:
It works! Excellent. :-)
Thank you.

closed 2016-11-05 by stevearc

thecxhacker, 2018-12-30:
Hi there, I started exploring vim just over the last few days. The last time I used vim was in school almost 17 years ago. Lol. Anyway, I recently had the suddenly spark/inspiration of wanting to pick up Vim again and I thought, "Why not try programming Arduino in Vim instead of the Arduino IDE?" ***Just me feeling adventurous***

Anyway, I am currently using my Raspberry Pi as the development platform (MacBook went down recently so I'm trying to use the opportunity to learn new things about RPI and perhaps try to maximize it's potential).

I have set up my Vim like a powerful universal IDE, but one last thing I am stuck with is setting up the Plugin for Arduino.

**Environment Details:**
===========================
Raspbian Linux

Where the directories are located:
Packages & Boards :package: :
/home/pi/.arduino15
 |
 |-- packages/
       |-- adafruit/
       |-- arduino/
       |-- chipKIT/
 |-- staging

Main Application Dir: 
/home/pi/arduino-1.8.8
 |
 |-- examples
 |-- hardware
 |-- java
 |-- lib
 | -- ...

Arduino CLI:
/usr/local/bin/arduino-cli (I followed the official Arduino CLI setup, even tried running the commands from my Konsole Terminal and it works.)

According to your Plugin's configurations, how should I set the values? When I leave everything as its defaults, it says it cannot find the g:arduino_dir.

And when I have set the g:arduino_dir configuration, when I launch the ***.ino** file and use the command **:ArduinoChooseBoard**, I get a message saying "==NO ENTRIES==".

I would really appreciate the help.

I believe putting this question here might be helpful to many youths or makers, or basically anyone that has a Raspberry Pi, wants to hack on the Arduino, and like me..crazy enough to want to use Vim when I have the IDE on the desktop. :P

stevearc, 2019-01-14:
Huh, that's odd...it should have been able to detect those settings automatically (unless maybe the arduino command wasn't in your PATH?). Glad you got it working!

WuerfelDev, 2023-06-19:
For me on Mac I had to set:

    vim.g.arduino_dir = '/Applications/Arduino.app/Contents/Java/'