stevearc/vim-arduino #5: Commands not pausing

palmenros, 2017-05-07:
I am currently using neovim v0.2.0 and when I type a command like :ArduinoVerify there is an output but it closes as it finished. However if I use make, it prompts at the end HIT ENTER TO CONTINUE. That's OK for ArduinoVerify, but I cannot use make for the rest of the commands.
Is there any way of pausing them?

stevearc, 2017-05-10:
Yeah that was done intentionally at the time because it seemed better. I'll take a look at adding another config var or some other reasonable alternative that'll get you the behavior you need.

stevearc, 2017-05-13:
A brief correction to my earlier statement: when building/uploading/etc the intended behavior is that the command completes and you see the `Press ENTER or type command to continue` on the shell, which then takes you back to vim. Removing the `:silent` before the `exe` (which you're essentially doing when calling `:make` directly instead of `:ArduinoVerify`) will add an *additional* prompt upon returning to vim that forces you to hit `ENTER` again before resuming. At least, that is the case in my vim 7.4. I suspect that the behavior is somehow different in neovim 0.2.0, so I'm going to install it and mess around until I find a solution that works in both.

stevearc, 2017-06-04:
I believe this should be fixed for all commands, but give the upload & serial commands a look and make sure they're functioning as expected for you

commit d8506d2b (2017-06-04): Fix #5: Commands not pausing in neovim
closed 2017-06-04 by stevearc