stevearc/vim-arduino #7: Conflict with processing

palmenros, 2018-02-10:
Nowadays, arduino only uses `.ino` extension, and `.pde` extension is reserved for processing.
I have the plugin `sophacles/vim-processing` installed, which provides syntax highlighting and makes and runs processing sketches.
The filetype detection works perfectly, and `.pde` files are detected to be processing files.
When vim-arduino isn't installed, running `:make` builds and runs processing sketches (like it should). However when vim-arduino is installed, it also detects the file to be a processing file, but when running `:make`, it runs `:ArduinoVerify`, which obviously fails.
Could you make an option to disable ArduinoVerify on `.pde` files?
Thank you

stevearc, 2018-02-11:
Yep! Pushed a small change. Let me know if you still see any issues

closed 2018-02-11 by stevearc

stevearc, 2018-02-13:
The problem is that you still have a ftdetect script somewhere (could be `vim-polyglot`?) that is setting the filetype to be arduino and the value of `makeprg` is maintained even if something later sets the filetype to be processing. First you should verify that the final filetype is correct. A `:se ft` should display `filetype=processing`. If that is correct, then all you have to do is tell vim to use the correct compiler after it lands on the processing filetype. Create a file in `.vim/ftplugin/processing.vim` and add the line `compiler processing`. That should configure it to use the correct makeprg when a file is detected to be processing.

palmenros, 2018-02-13:
That worked perfectly. Thank you!

commit c6276cd2 (2018-02-11): Fix #7: Don't enable commands on *.pde files