stevearc/vim-arduino #9: Serial Port/Serial Monitor Not Working

kamenomagic, 2018-05-20:
Arduino IDE itself, everything works perfectly as expected.

In vim-arduino, it is able to upload to the board perfectly. Awesome stuff.

But, it is unable to execute the "ArduinoChoosePort" command and therefore the "ArduinoUploadAndSerial" doesn't work either, reporting:
```
Error detected while processing function arduino#ChoosePort:
line    7:
No likely serial ports detected!
```

It's strange that it is able to upload on the correct port but then can't display or allow choosing of the ports.

Any thoughts?

stevearc, 2018-05-23:
We're looking for the serial port based on a couple of glob patterns which you can find here: https://github.com/stevearc/vim-arduino/blob/c6276cd256a2a30192a985d695d436c5c8a439f1/autoload/arduino.vim#L55-L60

If you can find out what the port actually is, it's easy to set the `g:arduino_serial_port_globs` value in your vimrc to match it properly. I'd also be curious to know what the pattern is for your port to see if it's something I should add to the default list of globs.

kamenomagic, 2018-05-23:
Oh OK sounds good. I will let you know, ideally in the form of a pull
request. If I can't then I'll  just respond here for your help

kamenomagic, 2018-05-27:
#10 I promised a pull request so here it is

closed 2018-05-29 by stevearc