stevearc/vim-arduino #13: ArduinoSerial outputs  "executing job failed: No such file or directory "

gviscardi, 2018-11-05:
OS: Arch
VIm: 8.1.470
Arduino: 1.87
Set the path in my .vimrc
Build and upload works, but when it comes to connecting via serial, it gives the error " executing job failed: No such file or directory "
I have chosen the port and board with the vim-arduino commands
Serial connection works in Arduino and arduino-mk

stevearc, 2018-11-06:
If I had to guess, I'd say that you're missing the command we're using to make the serial connection. If you check out the docs, the default command is `screen`, but you can customize it
https://github.com/stevearc/vim-arduino/blob/86b1e7f0cc976c8a18f06788f37c86a7a26ccca8/doc/arduino.txt#L82-L87

So first thing to check would be if you have screen installed, and if you want to use a different command for the serial connection you should change the value of `g:arduino_serial_cmd`

gviscardi, 2018-11-09:
Perfect, thank you! I did not realize you used screen. I need to start reading the docs more.

closed 2018-11-09 by gviscardi