stevearc/vim-arduino #22: Reusing the same console window

leikoilja, 2020-01-17:
Hey, @stevearc.
Thank you so much for your plugin - super useful!

Just wondering if it's possible to reuse the same console window when you 'Upload' a sketch without creating new windows? Please see my attached gif for better understanding.

When using `tmux` it creates new panes. Thou, if i m not in `tmux` it simply creates new console windows within vim every time i upload. 

I dont know if it's worth mentioning, but i m on neovim

Thank you

stevearc, 2020-02-24:
Hi @leikoilja, the built-in tmux behavior is pretty rudimentary, but if you want to be able to send the command to the same tmux pane you can do so with [vim-slime](https://github.com/jpalardy/vim-slime). Support for vim-slime was added [in this pull request](https://github.com/stevearc/vim-arduino/pull/19).

If you're fine with the terminal behavior and just want to disable tmux, you can do
```
let g:arduino_serial_tmux = ''
let g:arduino_verify_tmux = ''
let g:arduino_upload_tmux = ''
```

leikoilja, 2020-02-28:
Awesome, thanks @stevearc, that's exactly what i was looking for. :+1:

closed 2020-02-28 by leikoilja