stevearc/vim-arduino #26: The commands are not editor commands

e3172, 2020-05-30:
If I do any of the commands from this plugin it gives:
E492: Not an editor command
I am using neovim on arch linux
arduino ide is installed via pacman

e3172, 2020-05-30:
I just needed to open a .ino file

closed 2020-05-30 by e3172

MilkJug1, 2023-12-30:
I'm having the same issue here to, running Neovim v0.9.4, I can't seem to use these commands when in an .ino file and or .cpp or .cc file, but when I'm either in my dashboard/empty buffer window I can use them. I'm using Lazy.vim as my package manager, would that be an issue, I saw #59 and the fact it was closed with a fix, is there something else going on here at play?

[image]
[image]
I am using alpha.nvim as my dashboard here, which might be affecting it. I don't know. I looked through `:help arduino` and didn't see anything at all that I could do. 

[image]
When running inside of a .ino file, 
[image]
[image]
I for some reason have autocomplete for these command when I have no open buffer.

Sorry for commenting on a closed issue, I saw this was very similar to mine and thought it would be better to ask it here.

MilkJug1, 2023-12-30:
Looking through the code of this plugin, it looks like there is no support with .cc files, which that not working its fair, though I don't know whats causing the issue between .ino files and .cpp files.

AlexEscalante, 2024-01-22:
I have the very same problem that you have @MilkJug1. I am using LazyVim. I can use the commands only when no file is open. Some condition is reversed, maybe?

MilkJug1, 2024-01-25:
> I have the very same problem that you have @MilkJug1. I am using LazyVim. I can use the commands only when no file is open. Some condition is reversed, maybe?

It does sound like it, though it doesn't make a lot of sense on why that would be. Maybe something to do with the way lazy.nvim handles plugin loading? I might poke around the source code and see if it's just some check not working properly or something.

AlexEscalante, 2024-01-25:
It would be nice if you could have a look… I am still very new to Lua and
NeoVim.

Alex Escalante

Web & Mobile Development For Hire
http://audelabs.com

MilkJug1, 2024-02-04:
> It would be nice if you could have a look… I am still very new to Lua and NeoVim. Alex Escalante Web & Mobile Development For Hire http://audelabs.com
> […](#)

Okay, after a while and remembering to fix it. I think I did something to get it working. All I had to do was add the `ft = "arduino"` as an option for Lazy when installing it in my plugins.lua

What I mean: 
```lua
	{
		'stevearc/vim-arduino',
		lazy = false,
		ft = "arduino",
	},
```

This seemingly fixed it, well I think at least. I do have this setup in my lsp.lua config which uses mason, heres what that looks like, since that could also be what fixed it. 

```lua
require('mason').setup()
require('mason-lspconfig').setup_handlers({ 
		 --- Omitting the other parts of my lsp config, since thats not needed, just the arduino part.

	["arduino-language-server"] = function()
		require('lspconfig').arduino_language_server.setup {
			on_attach = on_attach,
			capabilities = capabilities,
			cmd = {
			'arduino-language-server'
		}
	}
end
})
```
Try this, it might fix it.  @AlexEscalante