stevearc/vim-arduino #28: Processing can only open its own sketches

RoToRx88, 2020-07-21:
- I just installed the plugin with the newest version using vim Plug.
- I installed the newest version of the arduino IDE currently available on ubuntu 20 LTS. (arduino IDE version: 2: 1.0.5+dfsg2-4.1)
- I then configured my board, my port, and my programmer that are corret. (i checked the value with the one I use in the arduino IDE that works fine)
- I created a folder named after my project, and created a simple file with the same name as the folder, and opened it in vim.

Using all of the above, I can't verify nor upload a sketch using the vim-arduino plugin.
I have a gtk error dialog telling me:
> processing can only open its own sketches and other files ending in .ino or .pde

Then it opens the arduino IDE with my sketch in it.
From the arduino IDE if I perform an upload it works straight.

Do you have any idea why the plugin can't upload the sketch ?

stevearc, 2020-07-27:
I think that version of the arduino IDE is massively out of date. The current latest version is [1.8.13](https://www.arduino.cc/en/main/software), and support for the CLI methods this project uses wasn't added until 1.5. I'd recommend downloading the newest version of the IDE from their website and trying again.

RoToRx88, 2020-07-28:
How well... My bad. Sorry for bothering you with this. I'll investigate and re-open this issue if needed though I don't think I'll have to.
I just wonder why ubuntu doesn't put a big warning when installing arduino through apt as a deprecated version.
Anyway thanks for your help and the plugin !

closed 2020-07-28 by RoToRx88