stevearc/vim-arduino #31: Compilation fails when using multiple source files

darksworm, 2020-08-12:
**User story:** When writing more complicated arduino sketches, I want to split the code up into multiple files.

**Issue**: Currently if I try to use `include` to inclue another source file, the compilation fails.

**How to recreate**

Create two files in the same directory:
```cpp
// test.c.ino
#include "test.h"

void setup() {}
void loop() {}
```

```cpp
// test.h
// this file is empty
```

Try to compile & upload the sketch and this error will appear:
```
test.c:1:10: fatal error: test.h: No such file or directory
 19  #include "test.h"
 20     ┆   ┆ ^~~~~~~~
 21 compilation terminated.
 22 exit status 1
```

This code, however, compiles and uploads successfully when using the arduino-ide.

darksworm, 2020-08-12:
Upon some further inspection this seems to be an issue with the arduino upload command.

```bash
arduino --upload --board arduino:avr:uno --port /dev/ttyUSB0 --pref programmer=arduino:avrispmkii --pref build.path=./build test.ino
```
results in the same error, where as this works (after changing `test.h` to `test.cpp`):
```bash
arduino-cli compile -b arduino:avr:uno . -v --build-path `pwd`/build && arduino-cli upload -p /dev/ttyUSB0 --fqbn arduino:avr:uno
```

darksworm, 2020-08-12:
Some time later... I don't know what I did, but it suddenly started working...

darksworm, 2020-08-16:
It has been worked flawlessly ever since. It was probably a peculiarity with my system.

stevearc, 2020-08-16:
Glad to hear it's working! LMK if you ever manage to repro

darksworm, 2020-10-15:
Just managed to reproduce it - the issue occurs when the main `ino` file is in a directory that is named differently than it.

In my case - my main source file is named `wul.ino`, so it expects that it's parent directory is called `wul`.

See attached gifs for reproduction steps & proof:
[image]
compiling `wul.ino` from a directory named `src` - does not work.

[image]
compiling `wul.ino` from a directory named `wul` - works.

I'm sure that this issue is isolated to the directory naming, but just in case, the source code I'm trying to compile is availlable [here](https://github.com/darksworm/wake-up-light/tree/ae3137ec975296fba46982e92c3ad11f03d2bfe6).

stevearc, 2020-10-19:
Unfortunately this seems to be a limitation of the Arduino toolchain. There has been some discussion and it's possible that this requirement will be lifted in the future https://github.com/arduino/arduino-cli/issues/948

grammoboy2, 2020-11-05:
I can only use one tab in nvim ifor Arduino it seems indeed. When I open another tab with a other file to edit, it still tries to compile the first file, even when I've closed that file.

stevearc, 2021-05-18:
Looks like the arduino-cli enhancement was declined. Seems like this is just the way Arduino works for the foreseeable future.

closed 2021-05-18 by stevearc