stevearc/vim-arduino #32: Arduino boards missing from list

T-vK, 2020-09-03:
I'm on x64 Linux (Fedora) with Arduino IDE 1.8.5, vim 8.2 and the latest master of vim-arduino.
For some reason when i list the boards it only shows ESP8266 and ESp32 boards, but no Arduino boards:
[image]

In the IDE all boards show up:
[image]

Any ideas what's wrong here?

stevearc, 2020-09-07:
The code for finding the board definitions is here:

T-vK, 2020-09-08:
It's `/etc/arduino/boards.txt` for me.

``` bash
$ ls -a /etc/arduino/
.  ..  boards.txt  preferences.txt  programmers.txt
```

``` bash
$ ls -l ~/.arduino15/packages/*/hardware/**/boards.txt
/home/fedora/.arduino15/packages/esp32/hardware/esp32/1.0.2/boards.txt
/home/fedora/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/boards.txt
```

``` bash
$ which arduino
/usr/bin/arduino
```

``` bash
ls "$(which arduino)/../hardware/**/boards.txt"
ls: cannot access '/usr/bin/arduino/../hardware/**/boards.txt': Not a directory
```

Seems like

https://github.com/stevearc/vim-arduino/blob/c26b81657229024740c5f7ea81acf8d9fbf72cdd/autoload/arduino.vim#L178

and

https://github.com/stevearc/vim-arduino/blob/c26b81657229024740c5f7ea81acf8d9fbf72cdd/autoload/arduino.vim#L216

would have to be adjusted as well.

stevearc, 2020-09-09:
/usr/bin/arduino is probably a symlink. The actual install location is likely `/usr/local/share/arduino-1.8.5` or similar.

This is an interesting difference because the current board locations have the package & arch in their paths, and it's required to pass that in to the command line e.g. `arduino --verify --board arduino:avr:gemma --pref programmer=arduino:usbtinyisp`

Would it be possible for you to attach the contents of your `boards.txt` and `programmers.txt` files (or link to a paste), as well as providing a CLI invocation that does a `--verify`? I'm curious if the `package:arch` is required for you and, if it is, where that data comes from if not the filepath.

T-vK, 2020-09-09:
`/usr/bin/arduino` is not a symlink and `/usr/local/share/arduino*` does not exist on my system.

``` bash
$ arduino --verify ~/Projects/misc/ArduinoDoorbell/ArduinoDoorbellTransmitter/ArduinoDoorbellTransmitter.ino 
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...
Sketch uses 22036 bytes (68%) of program storage space. Maximum is 32256 bytes.
Global variables use 1733 bytes (84%) of dynamic memory, leaving 315 bytes for local variables. Maximum is 2048 bytes.
Low memory available, stability problems may occur.
```

[boards.txt](https://github.com/stevearc/vim-arduino/files/5193484/boards.txt)
[programmers.txt](https://github.com/stevearc/vim-arduino/files/5193485/programmers.txt)

stevearc, 2020-09-09:
Your boards.txt file looks close to identical to the one I have in /usr/local/share. I dug through the arduino source a bit and it seems like they always expect to search for boards.txt files under a `/hardware` directory, so I'm not sure what's going on. I've added a workaround for now that hardcodes `/etc/arduino` and assumes a default package/arch for boards and programmers found there. Let me know if that does the trick

T-vK, 2020-09-10:
Thank you. I have pulled the latest master, but now I get the following error when opening an ino file:

```
Error detected while processing function arduino#InitializeConfig[49]..arduino#ReloadBoards[2]..arduino#GetArduinoDir:
line   11:
E605: Exception not caught: Could not find arduino directory. Please set g:arduino_dir
Error detected while processing /home/fedora/.vim/pack/vendor/start/vim-arduino/ftplugin/arduino.vim:
line    7:
E171: Missing :endif
Error detected while processing function <SNR>8_LoadFTPlugin:
line   17:
E170: Missing :endfor
Press ENTER or type command to continue
```

stevearc, 2020-09-10:
The weird thing is that you weren't getting that before. It happens in `arduino#GetArduinoDir()`, which was being called inside of `arduino#GetBoards()` before the change. Regardless, I've pushed up another change that does a best-effort search for all the directories it can find but doesn't error out if anything is missing. You can confirm which directories are being searched with the `:ArduinoInfo` command.

T-vK, 2020-09-10:
Nice, now I was able to select my board.
I wasn't able to compile using `:ArduinoVerify` though. It throws a Java error:

```
  1 Loading configuration...                                                                                                                                                                                                                   
  2 Initializing packages...                                                                                                                                                                                                                   
  3 Preparing boards...                                                                                                                                                                                                                        
  4 Verifying...                                                                                                                                                                                                                               
  5 java.lang.NullPointerException                                                                                                                                                                                                             
  6         at processing.app.helpers.FileUtils.listFiles(FileUtils.java:348)                                                                                                                                                                  
  7         at processing.app.Sketch.listSketchFiles(Sketch.java:108)                                                                                                                                                                          
  8         at processing.app.Sketch.<init>(Sketch.java:54)                                                                                                                                                                                    
  9         at processing.app.Base.<init>(Base.java:381)                                                                                                                                                                                       
 10         at processing.app.Base.main(Base.java:135)
```

stevearc, 2020-09-10:
This is starting to be more related to the arduino tool itself. I'd recommend using `:ArduinoInfo` to get the verify command and run it yourself in the terminal. If it works, then there's something up with the terminal environment in vim. If it doesn't, then we either need to change the generated command or there's an issue with the arduino CLI itself

T-vK, 2020-09-16:
Sorry for the late response. I only get the afore mentioned error when I run `:ArduinoVerify` in vim. 
`:ArduinoInfo` returns this for me:

```
Board         : arduino:avr:pro:cpu=8MHzatmega328
Programmer    : arduino:usbtinyisp
Port          : none
Baud rate     : 9600
Hardware dirs : /home/fedora/.arduino15/packages/esp8266/hardware/esp8266/2.6.3, /home/fedora/.arduino15/packages/esp32/hardware/esp32/1.0.2, /etc/arduino
Verify command: arduino --verify --board arduino:avr:pro:cpu=8MHzatmega328 --pref programmer=arduino:usbtinyisp --pref build.path='/home/fedora/Projects/misc/ArduinoDoorbell/ArduinoDoorbellTransmitter/build' --verbose-upload '/home/fedora/Projects/misc/ArduinoDoorbell/ArduinoDoorbellTransmitter/ArduinoDoorbellTransmitter.ino'
```

When I run the verify command manually I get no errors:

``` bash
$ arduino --verify --board arduino:avr:pro:cpu=8MHzatmega328 --pref programmer=arduino:usbtinyisp --pref build.path='/home/fedora/Projects/misc/ArduinoDoorbell/ArduinoDoorbellTransmitter/build' --verbose-upload '/home/fedora/Projects/misc/ArduinoDoorbell/ArduinoDoorbellTransmitter/ArduinoDoorbellTransmitter.ino'
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...
Sketch uses 22036 bytes (71%) of program storage space. Maximum is 30720 bytes.
Global variables use 1733 bytes (84%) of dynamic memory, leaving 315 bytes for local variables. Maximum is 2048 bytes.
Low memory available, stability problems may occur.
```

stevearc, 2020-09-17:
On vim 8, all we do is pass that command in directly to the terminal. ArduinoVerify is just calling a function
https://github.com/stevearc/vim-arduino/blob/dbb111557ba4b881f27a2f97a9961e794d907a39/ftplugin/arduino.vim#L22
Here's where we generate and run the command:
https://github.com/stevearc/vim-arduino/blob/dbb111557ba4b881f27a2f97a9961e794d907a39/autoload/arduino.vim#L407-L415
And here's where we set `s:TERM`:
https://github.com/stevearc/vim-arduino/blob/dbb111557ba4b881f27a2f97a9961e794d907a39/autoload/arduino.vim#L14-L17

I have no idea why running the command from vim's terminal would produce different output from running it directly on the command line. See if you can get a minimal repro just running commands by hand in `:terminal` and file it on the main vim repo. Maybe someone that knows more about vim internals will be able to help.

T-vK, 2020-09-17:
When I run `:terminal` in vim and then run the long `arduino --verify ...` command in that, it runs just fine. It only fails when I run `:ArduinoVerify`. :/

stevearc, 2020-09-17:
I have very limited ability to debug this since I don't have access to a case that reproduces this behavior. If running it in `:terminal` works then maybe it's an escaping issue, or maybe it's a problem with `exec`. The entirety of the function is 7 lines.
https://github.com/stevearc/vim-arduino/blob/dbb111557ba4b881f27a2f97a9961e794d907a39/autoload/arduino.vim#L407-L415

You should be able to easily track what it's doing and replicate it with raw vim commands until you repro the error. That will help us zero in on exactly which step is making the difference.

T-vK, 2020-09-17:
I don't know. I tried adding echo calls inside that arduino#Verify function, but it didn't appear in the output, even if I place it at the very beginning of the function.

stevearc, 2020-09-17:
Echo prints very transient messages. You can use `echom` instead, which will save it to the message history which you can later read with `:messages`. If you still don't see it, delete the function and see if it still runs to make sure that you're actually editing the right piece of code

T-vK, 2020-09-18:
Okay the function is definitely getting called. I'm really struggling with this because the language seems to be very different from bash, but using:

``` Bash
function! arduino#Verify() abort
  let cmd = arduino#GetArduinoCommand("--verify")
  echom "----cmd contents start----"
  echom(cmd)
  echom "----cmd contents end----"
  if g:arduino_use_slime
    call slime#send(cmd."\r")
  else
    echom "----s:TERM . cmd----"
    echom s:TERM . cmd 
    exe s:TERM . cmd 
    echom "----EXE FINISHED----"
  endif
  echom "----v:shell_error----"
  echom v:shell_error
  echom "----DONE----"
  return v:shell_error
endfunction
```

I get the following the the `:messages`:

```
----cmd contents start----
arduino --verify --board arduino:avr:pro:cpu=8MHzatmega328 --pref programmer=arduino:usbtinyisp --pref build.path='/home/fedora/Projects/misc/ArduinoDoorbell/ArduinoDoorbellTransmitter/build' --verbose-upload '/home/fedora/Projects/misc/Ar
duinoDoorbell/ArduinoDoorbellTransmitter/ArduinoDoorbellTransmitter.ino'
----cmd contents end----
----s:TERM . cmd----
terminal! arduino --verify --board arduino:avr:pro:cpu=8MHzatmega328 --pref programmer=arduino:usbtinyisp --pref build.path='/home/fedora/Projects/misc/ArduinoDoorbell/ArduinoDoorbellTransmitter/build' --verbose-upload '/home/fedora/Projec
ts/misc/ArduinoDoorbell/ArduinoDoorbellTransmitter/ArduinoDoorbellTransmitter.ino'
----EXE FINISHED----
----v:shell_error----
0
----DONE----
```

And when I run that command stored in `cmd` manually I get no errors.

stevearc, 2020-09-19:
The main thing that's different between this execution and when you run the command manually is that this is running through `exec`. When you run 
```viml
:exec "terminal! arduino --verify --board arduino:avr:pro:cpu=8MHzatmega328 --pref programmer=arduino:usbtinyisp --pref build.path='/home/fedora/Projects/misc/ArduinoDoorbell/ArduinoDoorbellTransmitter/build' --verbose-upload '/home/fedora/Projects/misc/ArduinoDoorbell/ArduinoDoorbellTransmitter/ArduinoDoorbellTransmitter.ino'"
```
that will either reproduce the issue or make me question my sanity.

I don't see anything obviously wrong with the escaping or quoting, but it's possible that there's an issue there. One way to check this would be to replace the `arduino` command with `test.sh`, and then add a `test.sh` script to your PATH:
```sh
#!/bin/bash
env # Print all environment variables
echo "Arduino cmd: $(which arduino)"
echo "---Arguments---"
for arg in "$@"; do
 echo "Arg: $arg"
done
```
Then you can try running the command by hand or running the command via the plugin and see if the script produces different output.

Another thing you can try to see if it behaves differently is running with `:!arduino --verify ...` instead of using `:terminal`, or by using `:call system(s:TERM . cmd)`

T-vK, 2020-09-21:
The test.sh script gives me the following output:

```
Arduino cmd: /usr/bin/arduino                                                                                                                                                                                                                  
---Arguments---                                                                                                                                                                                                                                
Arg: --verify                                                                                                                                                                                                                                  
Arg: --board                                                                                                                                                                                                                                   
Arg: arduino:avr:pro:cpu=8MHzatmega328                                                                                                                                                                                                         
Arg: --pref                                                                                                                                                                                                                                    
Arg: programmer=arduino:usbtinyisp                                                                                                                                                                                                             
Arg: --pref                                                                                                                                                                                                                                    
Arg: build.path='/home/fedora/Projects/misc/ArduinoDoorbell/ArduinoDoorbellTransmitter/build'                                                                                                                                                  
Arg: --verbose-upload                                                                                                                                                                                                                          
Arg: '/home/fedora/Projects/misc/ArduinoDoorbell/ArduinoDoorbellTransmitter/ArduinoDoorbellTransmitter.ino'
```

The only diffrence in env vars between running with `:exec` and with `:!` is that `:!` gives me a few additional env vars:

```
COLORS=256
ROWS=15
LINES=15
COLUMNS=239
VIM_TERMINAL=802
```

But otherwise they atre identical.

`:!arduino --verify ...` works fine.

For `:call system(s:TERM . cmd)` I get:

```
E121: Undefined variable: s:TERM
E116: Invalid arguments for function system
```

Any ideas?

stevearc, 2020-09-21:
#34 reported a similar but slightly different error. Could you try the fix I pushed for that one? I'm hoping that it's the same root cause.

T-vK, 2020-09-22:
Thank you so much! It works now. :)

closed 2020-09-22 by T-vK