stevearc/vim-arduino #33: Not working with vim under Ubuntu 18.04

Eric97, 2020-09-17:
After installing the plugin with `plug`, all the arduino commands cannot be invoked either in Vim or NeoVim. The default `arduino_cmd` is installed at `/usr/local/bin/arduino`.

stevearc, 2020-09-17:
When you say "cannot be invoked", what is actually happening? And please post the output of `:ArduinoInfo` if you can.

Eric97, 2020-09-17:
`:ArduinoInfo` and the other `:Ardu*` commands are not recognized as valid editor commands. 
`:help arduino` can display the help document though.

stevearc, 2020-09-17:
Ah, the commands are defined in the ftplugin file https://github.com/stevearc/vim-arduino/blob/master/ftplugin/arduino.vim

It's possible that the files you're editing aren't properly recognized. The ftdetect rules live here: https://github.com/stevearc/vim-arduino/blob/master/ftdetect/arduino.vim

Try doing a `:se ft?` to check the detected filetype. If it's not set to `arduino` then that would explain it, and then we can figure out why the detection isn't working.

Eric97, 2020-09-18:
Thanks. After opening the `.ino` file in vim, then the commands can work with the correct file type detected!

closed 2020-09-18 by Eric97