stevearc/vim-arduino #34: :ArduinoVerify fails

descrip, 2020-09-21:
Hi! Just installed the plugin. My IDE is 1.8.13, and Vim is at 8.1.1401. I coudn't get `:ArduinoVerify` to run --- it gives this error:

```
Picked up JAVA_TOOL_OPTIONS: 
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...
java.io.IOException: No valid code files found
        at processing.app.Sketch.listSketchFiles(Sketch.java:117)
        at processing.app.Sketch.<init>(Sketch.java:54)
        at processing.app.Base.<init>(Base.java:423)
        at processing.app.Base.main(Base.java:150)
```

The command it's attempting to run is:

```
arduino --verify --board arduino:avr:mega:cpu=atmega2560 --port /dev/ttyACM0 --pref programmer=arduino:avrisp --pref build.path='/home/jeff/Arduino/RGB_LED/build' --verbose-upload '/home/jeff/Arduino/RGB_LED/RGB_LED.ino'
```

[Looking at the :ArduinoVerify command](https://github.com/stevearc/vim-arduino/blob/dbb111557ba4b881f27a2f97a9961e794d907a39/autoload/arduino.vim#L407), it seems that we can run with `TERM` being either `:!` or `:terminal!`. I should be using `:terminal!`. Running the command using `:! $VERIFY_COMMAND` works perfectly fine (and so does running it in a separate terminal). However, if I run `:terminal! $VERIFY_COMMAND`, it gives the same error.

I think something is incorrect about how `:terminal!` is being used. Thanks!

stevearc, 2020-09-21:
I was able to reproduce this by using vim 8. Turns out that the quoting rules for a command passed to `:terminal` are not the same as for `:!`. See https://github.com/vim/vim/issues/4288
This was causing us to send command arguments with a literal `'` to the `arduino` command. Arduino couldn't find a file with a literal `'`, so it barfed.

Should be fixed in 1a4fd1576550b1b6fd1d90ea7515ca209f09fe67

FWIW, I switched to Neovim over a year ago and it's fantastic. Would highly recommend 🙃

descrip, 2020-09-21:
Wow, thanks for the quick fix! The command works after updating the plugin.

I decided to stay on Vim since it seemed more stable than shiny new Neovim. After seeing that bug, I'm having second thoughts...

Thanks again!

closed 2020-09-21 by descrip