stevearc/vim-arduino #35: Command to change the debug serial baudrate

mroavi, 2020-12-10:
It would be nice to have a command that could be used in a mapping to update the default debug serial baudrate.

stevearc, 2020-12-11:
Do you mean a command to manually specify the baud rate? Because that's done by simply setting `let g:arduino_serial_baud = 9600`
https://github.com/stevearc/vim-arduino/blob/1a4fd1576550b1b6fd1d90ea7515ca209f09fe67/doc/arduino.txt#L104-L106

Or do you mean a command to do the automatic baud rate detection on the current file? There's no command for that currently but you could easily map the function call
https://github.com/stevearc/vim-arduino/blob/1a4fd1576550b1b6fd1d90ea7515ca209f09fe67/ftplugin/arduino.vim#L16-L18

mroavi, 2020-12-11:
Mmm... I mean a command of this type:

```
The main commands you will want to use are:

:ArduinoChooseBoard - Select the type of board from a list.
:ArduinoChooseProgrammer - Select the programmer from a list.
:ArduinoChoosePort - Select the serial port from a list.
:ArduinoVerify - Build the sketch.
:ArduinoUpload - Build and upload the sketch.
:ArduinoSerial - Connect to the board for debugging over a serial port.
:ArduinoUploadAndSerial - Build, upload, and connect for debugging.
:ArduinoInfo - Display internal information. Useful for debugging issues with vim-arduino.

```
So that I can invoke it with a key binding for convenience.

stevearc, 2020-12-11:
It's still not clear to me if what you want to do with your keybinding is set the baud rate to a value or perform the auto detection.

Assuming it's the auto detection, you can already do that by calling the function `arduino#SetAutoBaud()`
```vim
nnoremap <buffer> <leader>b :call arduino#SetAutoBaud()<CR>
```

mroavi, 2020-12-13:
I would like to have a mapping that has the exact same functionality as `let g:arduino_serial_baud = 9600`. I usually have several sketches open that use different debug serial baudrates, so it would be great to (1) be able to configure them with a mapping (and not having to look for my .vimrc, edit it, and source it) and 2. make this variable buffer specific, so that I could for example use 9600 with one sketch, and 115200 with another one.

stevearc, 2020-12-16:
Gotcha! That's a pretty easy add

commit aef4ab52 (2020-12-16): Fix #35: Add a command to set the serial baud
closed 2020-12-16 by stevearc