stevearc/vim-arduino #41: ArduinoSerial doesn't exit cleanly

sigmaSd, 2021-09-19:
Trying to run ArduinoSerial a second time will result in an error `port busy` because the screen command is still running.
 
The problem is the `screen` command, which doesn't exit after closing the serial panel. (see https://stackoverflow.com/questions/6301840/how-to-stop-a-screen-process-in-linux)

I'm working-around this by  `pkill screen; screen {port}` but I don't think that's an acceptable solution.

stevearc, 2021-09-21:
Instead of closing the window with the `screen` command buffer, try either ctrl+c to interrupt the screen command or `:bdelete` to delete the buffer. I think the problem you have is that you're closing the window but the terminal buffer (and its associated screen command) are still hanging around in the background (you can verify with `:ls!`).

sigmaSd, 2021-09-21:
`sceen` doesn't respond to `ctrlc`, I'll try `:bdelete` as you suggested.

Anyway I thought the issue is reproducible, I understand from what you're saying that its not the case so I'll just close the issue.

closed 2021-09-21 by sigmaSd