stevearc/vim-arduino #44: How to stop Serial Monitor

saschahoeche, 2021-11-16:
While the plugin works really great, i don't know how to stop/close the serial monitor after opening it and reading the serial stream.
Any pointers?

stevearc, 2021-11-17:
If you're using `screen` (the default, see `:h arduino_serial_cmd`) would think that ctrl+c should stop it. Is that not working for you?

saschahoeche, 2021-11-18:
I use the :ArduinoUploadAndSerial command or the :ArduinoSerial command. But Ctrl+C doesn't work on either. I am sure the fault is on my side but i can't seem to figure it out.

stevearc, 2021-11-18:
Ah, of course it's screen so you have to kill the screen session using it's escape key. By default this is bound to `<C-a><C-k>`

saschahoeche, 2021-11-24:
Cheers, that worked

closed 2021-11-25 by stevearc