stevearc/vim-arduino #45: ctags: no such file or directory

ziidonato, 2021-11-21:
On Opensuse Tumbleweed

output of :ArduinoVerify
` Picked up JAVA_TOOL_OPTIONS:                               
  1 Loading configuration...                       
  2 Initializing packages...          
  3 Preparing boards...                                          
  4 Verifying...                  
  5 fork/exec {runtime.tools.ctags.path}/ctags: no such file or directory
  6  
  7 [Process exited 1]`

`ctags --version
Exuberant Ctags Development, Copyright (C) 1996-2009 Darren Hiebert
  Addresses: <dhiebert@users.sourceforge.net>, http://ctags.sourceforge.net
  Optional compiled features: +wildcards, +regex`

stevearc, 2021-11-21:
This looks like an issue with the arduino command, not with this plugin. Run `:ArduinoInfo`, that should tell you what command is actually being run when you do `:ArduinoVerify`. Try running that command directly in the shell; I suspect it will fail with this error.

ziidonato, 2021-11-22:
> This looks like an issue with the arduino command, not with this plugin. Run `:ArduinoInfo`, that should tell you what command is actually being run when you do `:ArduinoVerify`. Try running that command directly in the shell; I suspect it will fail with this error.

Apparently it is. Sorry for wasting your time

Great plugin though, use it all the time

closed 2021-11-22 by ziidonato