stevearc/vim-arduino #46: E716: Key not present in Dictionary: "boards"

mattbrowley, 2021-12-09:
**Describe the bug**
When I run :ArduioChooseBoard, it throws the error in the title. I have arduino installed, and I have pointed to the directory in my ftplugin/arduino.vim with 
let g:arduino_dir = "/usr/local/share/arduino"
I have checked that directory, and boards.txt is present in /usr/local/share/arduino/avr/

**System information**
- OS: linux (Pop!_OS 21.04x64)
- Vim: nvim v0.6.0
- Arduino: Arduino: 1.8.16
- CLI version: arduino-cli  Version: 0.20.1 Commit: abb21449 Date: 2021-11-29T11:34:58Z
- ArduinoInfo: 
Board         : arduino:avr:uno
Programmer    : arduino:usbtinyisp
Port          : none
Baud rate     : 9600
Hardware dirs : /usr/local/share/arduino/hardware/arduino/avr
Verify command: arduino-cli compile -b arduino:avr:uno -P arduino:usbtinyisp --build-path "/home/matthewrowley/Arduino/CD_Rob
ot/build" -v "/home/matthewrowley/Arduino/CD_Robot/CD_Robot.ino"

**To Reproduce**
Steps to reproduce the behavior:
1.  load vim-arduino using vim-plug in .vimrc
2.  open any .ino file
3. run command :ArduinoChooseBoard

**Expected behavior**
With no argument, this command should give me a list of all installed boards and let me choose one

**Working command**
This bug is not regarding verify/upload issues

**Screenshots**
[image]

**Additional context**
I installed arduino with the install.sh script rather than using the Pop!_OS package manager because I wanted the latest stable version. I installed the binary to /usr/local/bin. This didn't seem to put the boards.txt file anywhere at all, such as in ~/.arduino15/, so I looked at the output of "verify" within the Arduino IDE and noted that it was referencing the original downloaded folder where install.sh was located. So, I removed arduino from /usr/local/bin, manually moved the downloaded folder from "Downloads" into a more permanent home at /usr/local/share, and then installed again to /usr/local/bin from there. Now the IDE output shows many references to that folder, and I felt confident to proceed to include let g:arduino_dir = "/usr/local/share/arduino" in my ftplugin/arduino.vim

I included some other sane defaults based on information from the vim-arduino help. Below is the entirety of my ftplugin/arduino.vim file right now:

let g:arduino_cmd = "/usr/local/bin/arduino"
let g:arduino_home_dir = "~/.arduino15"
let g:arduino_dir = "/usr/local/share/arduino"
let g:arduino_build_path = "{project_dir}/build"
let g:arduino_run_headless = 1
"let g:arduino_board = 'arduino:avr:pro'
"let g:arduino_programmer = 'arduino:avrispmkii'
let g:arduino_board = 'arduino:avr:uno'
let g:arduino_programmer = 'arduino:usbtinyisp'
let g:arduino_serial_cmd = 'screen {port} {baud}'
let g:arduino_auto_baud = 1  " Searches script for Serial.begin() for baudrate
" let g:arduino_serial_port = '/dev/ttyUSB0'

nnoremap <buffer> <leader>am :ArduinoVerify<CR>
nnoremap <buffer> <leader>au :ArduinoUpload<CR>
nnoremap <buffer> <leader>ad :ArduinoUploadAndSerial<CR>
nnoremap <buffer> <leader>ab :ArduinoChooseBoard<CR>
nnoremap <buffer> <leader>ap :ArduinoChooseProgrammer<CR>

Thank-you for your help!

stevearc, 2021-12-09:
This is a *fantastic* bug report, thank you.

So in your case vim-arduino will mostly be making use of the `arduino-cli` instead of trying to read files from the arduino install directory. The exact error is happening [on this line](https://github.com/stevearc/vim-arduino/blob/6bb4c41477c53b01377e696559f2399401da29f7/autoload/arduino.vim#L231) where we try to pull the boards out of some json output. From that I assume that the output of the command is either empty or in a different format than we expect. I thought it might be the version of `arduino-cli`, but I updated to your version and my output still looks like I expect.

Could you run `arduino-cli board listall --format json` and paste the output here?

mattbrowley, 2021-12-09:
It just returns an empty {}

I did try to run down obvious user errors before posting, so I'm glad the bug report is helpful.

Thank you for your quick response.

stevearc, 2021-12-09:
Okay, so this seems like an issue with the cli tool then. I haven't encountered this myself, but [some people are recommending](https://forum.arduino.cc/t/arduino-cli-on-rpi-cannot-find-boards/647948) a `arduino-cli core update-index` and a `arduino-cli core install arduino:avr`. Running `arduino-cli core list --all` should show you which boards are installed and which are available.

mattbrowley, 2021-12-09:
Now :ArduinoChooseBoard is giving me the  list as expected, but when I choose the Pro Mini I get:
[image]

mattbrowley, 2021-12-09:
Oh, and arduino-cli board listall --format json now gives a big json database which includes an entry for the Pro Mini

{
      "name": "Arduino Pro or Pro Mini",
      "fqbn": "arduino:avr:pro",
      "platform": {
        "id": "arduino:avr",
        "installed": "1.8.4",
        "latest": "1.8.4",
        "name": "Arduino AVR Boards",
        "maintainer": "Arduino",
        "website": "http://www.arduino.cc/",
        "email": "packages@arduino.cc"
      }

stevearc, 2021-12-09:
Great, that's progress! I'm not sure what could be causing the new error, but upon review I was doing the json decoding in a stupid way and I changed it to be significantly less stupid. Update and see if that makes a difference

stevearc, 2021-12-09:
Side note and shameless self promotion: I notice that you're using Neovim 0.6. If you'd like a better chooser interface (say, Telescope or a floating window), check out https://github.com/stevearc/dressing.nvim

mattbrowley, 2021-12-09:
Now it's a new error
[image]
:

stevearc, 2021-12-09:
Great! This is actually the underlying error that was masked by the previous error. This is caused by what I suspected at the very beginning: your arduino-cli version is newer than the one I used when I first added support for it and the API has changed. I updated the command args and luckily they seem to be backwards compatible with older versions of the cli.

Hopefully this should be the last of it!

mattbrowley, 2021-12-09:
You may already be working on this, but I am getting an error setting the programmer too. If it is a new problem, I can open a new issue if you want.
[image]

mattbrowley, 2021-12-09:
It looks like it all works! I verified a program, but I couldn't try uploading or opening a serial connection because I don't have a board on hand right now.
Thank-you! I've been wanting to do arduino in vim for a while now.