stevearc/vim-arduino #47: avrdude: usbdev_open(): did not find any USB device "usb" (0x03eb:0x2104)

meikse, 2022-01-15:
After verifying, by calling :ArduinoUpload I get the error:

```
         Using Port                    : usb
         Using Programmer       : stk500v2
avrdude: usbdev_open(): did not find any USB device "usb" (0x03eb:0x2104)

avrdude done.  Thank you.

Error during Upload: Failed programming: uploading error: exit status 1)
```

so the port is supposed to be something like 

` Using Port                    : /dev/ttyACM0 `

also in the ArduinoInfo, I get the following

``` 
Board           : arduino:avr:uno                                  
Programmer : avrispmkii
Port              : /dev/ttyACM0
Baud rate     : 9600
```
I tried to find the bug inside autoload/arduino.vim but I have no idea where you declare the "arduino-cli upload", since the `:echo arduino#GetCLICompileCommand('-u')` would not change the arduino-cli command.

Can you tell me how to correct this bug?? ..because your plugin seems to be very handy! nice work!

ps: compiling and uploading the sketch with arduino-cli works properly.

meikse, 2022-01-16:
I am sorry for the disturbance. by default the programmer `avrispmkii` is chosen. 
By simply changing it to `avrisp` with `:ArduinoChooseProgrammer` it runs flawlessly. 

No Bugs with vim-arduino
No Bugs with arduino-cli

..just me being tired and producing easy errors

closed 2022-01-16 by meikse

stevearc, 2022-01-16:
No worries! Glad you figured it out

meikse, 2022-01-16:
edit: also my closed answer is not properly correct.

it worked when I selected no programmer for the uno (obviously)

edit in init.vim i.e. : 
`let g:arduino_programmer = "" `