stevearc/vim-arduino #50: Feature request: use sketch.json for per sketch fqbn and part

xenos1984, 2022-02-22:
Based on the [sketch specification](http://arduino.github.io/arduino-cli/0.21/sketch-specification/), one may add a file named `sketch.json` to a sketch folder, so set the [board metadata](http://arduino.github.io/arduino-cli/0.21/sketch-specification/#metadata) (like fqbn and port) on a per sketch basis. This file can be created using the arduino-cli `board attach` command.

Would it be possible to use the settings from this file, if it is present, and the global `g:arduino_*` variables / `ArduinoChoose*` commands only in case that no such file exists? In particular:

* When compiling / uploading, give command line arguments `--fqbn` and `--port` (using vim-arduino global settings) only if not given in the `sketch.json`.
* Show board / port from `sketch.json` in the tag bar if present.

As another feature, would it be possible to call [`board attach`](http://arduino.github.io/arduino-cli/0.21/commands/arduino-cli_board_attach/) to save such metadata, after selecting it with `ArduinoChoose*`, per sketch instead of globally (maybe as a new vim-arduino command or `.vimrc` option)?

I think this would be a helpful improvement if one is working on several sketches which are compiled for different board types / uploaded on different ports, so that there is no need to change these settings when switching to another sketch. Also it would use a settings format which is compatible with using plain arduino-cli on the command line.

stevearc, 2022-02-24:
Good idea! We now read from the sketch file and use that to set the board & port values. When a sketch file is present those values are not passed to the `arduino-cli` command. There's also a new `ArduinoAttach` command that will run `arduino-cli board attach` in the background. Lastly, `ArduinoChooseBoard` and `ArduinoChoosePort` will update the value in the sketch.json file, if it exists.

stevearc, 2022-03-02:
I've updated it to read the sketch file (if one exists) on BufReadPost for .ino files. Does that suit your purpose?

xenos1984, 2022-03-02:
Works perfectly now, thanks a lot!

closed 2022-03-02 by xenos1984