stevearc/vim-arduino #54: Wrong baud rate, when trying to upload a sketch

GiorgosXou, 2022-12-29:
**Describe the bug**
Trying to upload any sketch, happens to use the wrong baud rate and fails to do so.

- *vim-arduino runs with `-b19200`:*
```terminal
"/home/xou/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/bin/avrdude" "-C/home/xou/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/etc/avrdude.conf" -v -V -patmega328p -carduino -P/dev/ttyUSB0 -b19200 "-Uflash:w:/home/xou/Desktop/xou/programming/hardware/arduino/uno/test0/build/test0.ino.hex:i"
```

- *While the arduino IDE runs with `-b115200`:*
```
/home/xou/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/bin/avrdude -C/home/xou/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/etc/avrdude.conf -v -patmega328p -carduino -P/dev/ttyUSB0 -b115200 -D -Uflash:w:/tmp/arduino_build_782589/test0.ino.hex:i 
```

---

**System information**
- OS: Arch Linux
- Vim: Neovim v0.8.1
- Arduino: 1.8.19
- CLI version:  0.29.0-arch
- ArduinoInfo: 
```terminal
Board         : arduino:avr:uno
Programmer    : arduinoasispatmega32u4
Port          : /dev/ttyUSB0
Baud rate     : 115200
Hardware dirs : /home/xou/.arduino15/packages/arduino/hardware/avr/1.8.6, /home/xou/.arduino15/packages/digistump/hardware/avr/1.6.7, /home/xou/.arduino15/packages/ATTinyCore/hardware/avr/1.5.2
Verify command: arduino-cli compile -b arduino:avr:uno -p /dev/ttyUSB0 -P arduinoasispatmega32u4 --build-path "/home/xou/Desktop/xou/programming/hardware/arduino/uno/test0/build" -v "/home/xou/Desktop/xou/programming/hardware/arduino/uno/test0/test0.ino"
```
*(Brand new installation of Arduino and vim-arduino)*

---

**To Reproduce**
Steps to reproduce the behavior:
1. Open a sketch like `Blink.ino`
2. Upload it with Arduino IDE
3. Upload it with vim-arduino

---

**Expected behavior**
It should had used the correct baud rate and get uploaded (?)

---

**Working command**
Running the command just with the correct baud rate just works and gets uploaded.

stevearc, 2022-12-29:
Have you tried setting `let g:arduino_serial_baud = 115200`?

GiorgosXou, 2022-12-31:
*(Sorry for the late response)* Yes, it doesn't work, it only changes the serial baud-rate that it is supposed to listen to, and not the one for uploading a sketch

stevearc, 2023-01-04:
Ah, sorry I misread that. According to the debug information, the "verify" command it's running is 
```
arduino-cli compile -b arduino:avr:uno -p /dev/ttyUSB0 -P arduinoasispatmega32u4 --build-path "/home/xou/Desktop/xou/programming/hardware/arduino/uno/test0/build" -v "/home/xou/Desktop/xou/programming/hardware/arduino/uno/test0/test0.ino"
```
That means that the upload command is going to just add a `-u` to it
```
arduino-cli compile -b arduino:avr:uno -p /dev/ttyUSB0 -P arduinoasispatmega32u4 --build-path "/home/xou/Desktop/xou/programming/hardware/arduino/uno/test0/build" -u -v "/home/xou/Desktop/xou/programming/hardware/arduino/uno/test0/test0.ino"
```

If you run that manually from the command line, do you see the same error? If so, is there any invocation of `arduino-cli` that you can get to work? I don't see any arguments to specify the upload baud rate.

GiorgosXou, 2023-01-05:
Not specifing aprogrammer also works:
```
arduino-cli compile -b arduino:avr:uno -p /dev/ttyUSB0 --build-path "/home/xou/Desktop/xou/programming/hardware/arduino/uno/test0/build" -u -v "/home/xou/Desktop/xou/programming/hardware/arduino/uno/test0/test0.ino"
```
```terminal
"/home/xou/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/bin/avrdude" "-C/home/xou/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/etc/avrdude.conf" -v -V -patmega328p -carduino "-P/dev/ttyUSB0" -b115200 -D "-Uflash:w:/home/xou/Desktop/xou/programming/hardware/arduino/uno/test0/build/test0.ino.hex:i"

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/home/xou/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/etc/avrdude.conf"
         User configuration file is "/home/xou/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/ttyUSB0
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 3
         Firmware Version: 4.4
         Vtarget         : 0.3 V
         Varef           : 0.3 V
         Oscillator      : 28.800 kHz
         SCK period      : 3.3 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: reading input file "/home/xou/Desktop/xou/programming/hardware/arduino/uno/test0/build/test0.ino.hex"
avrdude: writing flash (924 bytes):

Writing | ################################################## | 100% 0.17s

avrdude: 924 bytes of flash written

avrdude done.  Thank you.

Used platform Version Path
arduino:avr   1.8.6   /home/xou/.arduino15/packages/arduino/hardware/avr/1.8.6
```
Those work too:
```
arduino-cli compile --fqbn arduino:avr:uno
arduino-cli upload --fqbn arduino:avr:uno -p /dev/ttyUSB0 --verbose
```

stevearc, 2023-01-06:
Well, if it's the programmer that's causing the issue, can you `let g:arduino_programmer = ''` and try again? That should remove the programmer from the compile/upload command.

I'm away from my computer that has arduino things installed atm, but if I remember correctly I think it's awkward/not possible to _clear_ the programmer using `:ArduinoChooseProgrammer`, which is why you have to set the variable directly. If this works I'll tweak the command to make it easier to clear the programmer.

GiorgosXou, 2023-01-06:
Ah... there's a `-None-` option in select programmer that I've managed to somehow miss, I'm sorry, it uploads just fine if I do select the `-None-` option

closed 2023-01-06 by GiorgosXou