stevearc/vim-arduino #56: The plugin does not seem to work in Vim9

ubaldot, 2023-06-12:
**Describe the bug**
I tried a number of commands and I got several errors. 
For example:
`:ArduinoVerify` return 

`Error opening sketch: main file missing from sketch: /Users/ubaldot/Documents/arduino/arduino.ino`, 

`:ArduinoChooseBoard` return 

`E370: Could not load library /opt/homebrew/lib/liblua.dylib: dlopen(/opt/homebrew/lib/liblua.dylib, 0x009): tried: /opt/homebrew/lib/liblua.dylib (no such file) ... `

**System information**
- OS: Mac Os X 13.4
- Vim: 9.0.1276
- CLI version: arduino-cli  Version: 0.33.0 Commit: ca60d4b4 Date: 2023-05-29T15:18:18Z
- ArduinoInfo: 
- 
```
Board         : arduino:avr:uno
Programmer    : 
Port          : /dev/tty.usbmodem101
Baud rate     : 9600
Hardware dirs : /Users/ubaldot/Library/Arduino15/packages/arduino/hardware/avr/1.8.6
Verify command: arduino-cli compile -b arduino:avr:uno -p /dev/tty.usbmodem101 --build-path "/Users/ubaldot/Documents/arduino/build" -v "/Users/ubaldot/Documents/arduino/pippo.ino"
```

**To Reproduce**
Install MacVim 9.0.1276, open a .ino file and run any vim-arduino command.

**Expected behavior**
I can operate with my arduino uno board. 

**Screenshots**
[image]

[image]

[image]

stevearc, 2023-06-14:
I believe it should be fixed. Can you give it another try?

ubaldot, 2023-06-14:
Yes, now it does. :) Thanks!

closed 2023-06-14 by ubaldot