stevearc/vim-arduino #59: Commands not found after installing with lazy.

PlexSheep, 2023-10-25:
**Describe the bug**
The commands of the plugin cannot be found when installing it with the lazy package manager, regardless of how I load
the plugin.

```lua
	{
		"stevearc/vim-arduino",
		cmd = {
			"ArduinoAttach",
			"ArduinoChooseBoard",
			"ArduinoChooseProgrammer",
			"ArduinoChoosePort",
			"ArduinoVerify",
			"ArduinoUpload",
			"ArduinoSerial",
			"ArduinoUploadAndSerial",
			"ArduinoInfo",
		},
	},
```

This config should load the plugin when I call one of the commands listed above. It tries to do so, but the commands are not found.

> Command `ArduinoInfo` not found after loading `vim-arduino`

When I try to load the plugin as non lazy like below, the commands just do not exist.

```lua
	{
		"stevearc/vim-arduino",
		lazy = false,
	},
```

> E492: Not an editor command: ArduinoInfo

**System information**
- OS: Linux, Fedora 38 (KDE Plasma)
- Vim: NeoVim `v0.9.2` with lazy
- Arduino: 1.8.19 (as flatpak)
- CLI version: `arduino-cli  Version: 0.34.2 Commit: 963c1a76 Date: 2023-09-11T10:05:42Z`
- ArduinoInfo: Not appliciable

**To Reproduce**
Steps to reproduce the behavior:
1. Install the plugin with lazy
2. Try to use any command

**Expected behavior**
The commands are found in both lazy and non lazy mode

**Additional context**
I have installed the arduino application as a flatpak, but `arduino-cli` is installed natively through their provided install script. If I understand correctly, this should not be a problem, as the plugin uses `arduino-cli` if possible and regular `arduino` only as fallback. `arduino-cli` is available from my regular shell, `arduino` is not (only as flatpak).

stevearc, 2023-10-28:
This is probably because the commands were defined in a `ftplugin` file instead of a `plugin` file. This was for historical reasons, but I don't think it makes much sense anymore. I've changed the commands to be globally available.

closed 2023-10-28 by stevearc

PlexSheep, 2023-10-28:
I can confirm this issue is fixed in the latest update. Thank you!

GiorgosXou, 2024-02-27:
So... seems like it happend again, and the way I "solved" it *(or what i did for myself)* was:
```lua
{'stevearc/vim-arduino', lazy = false }, -- sudo pacman -S screen arduino-cli (and arduino?) | arduino-cli config init
```
```lua
local api  = vim.api
api.nvim_command('au BufRead,BufNewFile *.ino nnoremap <buffer> <leader>aa <cmd>call arduino#Attach()<CR>')
api.nvim_command('au BufRead,BufNewFile *.ino nnoremap <buffer> <leader>am <cmd>call arduino#Verify()<CR>')
api.nvim_command('au BufRead,BufNewFile *.ino nnoremap <buffer> <leader>au <cmd>call arduino#Upload()<CR>')
api.nvim_command('au BufRead,BufNewFile *.ino nnoremap <buffer> <leader>ad <cmd>call arduino#UploadAndSerial()<CR>')
api.nvim_command('au BufRead,BufNewFile *.ino nnoremap <buffer> <leader>ab <cmd>call arduino#ChooseBoard()<CR>')
api.nvim_command('au BufRead,BufNewFile *.ino nnoremap <buffer> <leader>ap <cmd>call arduino#ChooseProgrammer()<CR>')
```
and so I thought to share it here.