stevearc/vim-arduino #62: can't execute commands when file launched through file manager

windostechsuport, 2024-07-23:
**Describe the bug**
I can't execute any of the Arduino commands when a file is loaded from a file manager such as nerdtree or telescope, even though it is shown as loaded in lazy.nvim . Lazy loading is disabled.
If I open a file directly via the command line and enter any Arduino command everything works without an issue.

**System information**
- OS: [gnu/linux]
- Vim: [neovim 10.0-5]
- Arduino: [1.8.19]
- CLI version: [0.35.3]

**To Reproduce**
Steps to reproduce the behavior:
1. open nvim 
2. launch i.e NERDtree 
3. open any desired text file

**Expected behavior**
to be able to use vim-arduino regardless of launching method.

windostechsupport, 2024-07-24:
Sorry if this is not directly related to the 
[image]
question. But is there a way to prevent vim-arduino to open up a new tab for every buffer? everytime I upload something the bar gets littered with undesired tabs.

windostechsupport, 2024-07-24:
> Sorry if this is not directly related to the question. But is there a way to prevent vim-arduino from opening up a new tab for every buffer? everytime I upload something the bar gets littered with undesired tabs. ![Bildschirmfoto_2024-07-24_08-02-51](https://private-user-images.githubusercontent.com/176398765/351592140-e96cee32-ef56-4ae4-84aa-1a9d6db484e6.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjE4MDE3MzEsIm5iZiI6MTcyMTgwMTQzMSwicGF0aCI6Ii8xNzYzOTg3NjUvMzUxNTkyMTQwLWU5NmNlZTMyLWVmNTYtNGFlNC04NGFhLTFhOWQ2ZGI0ODRlNi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNzI0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDcyNFQwNjEwMzFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kNWU2NDM2MjVhZGE5MzkwMGM2ODViOGNmNjRhMzY0Y2JhYTQwOTlmNGMxNDM1YTUyNWQ1YjI5M2QwNDM4OTI3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.4up-gTaftt9w5EHfDOE-NVVkOhRxXBnw4IgB0zYXBNI)

Oh I noticed to actually kill the process you need to do :bd! instead of just :q

Langtude-Larson, 2024-12-04:
Currently having the same issue, the commands are not seen as "editor commands" if i open the .ino file through neo-tree or any other in-editor method. but if i open the file directly through terminal everything works as intended, bumping in hopes someone has a way to fix this

TheSillySmoker, 2025-11-22:
Bump! Same problem :(